ranenbg/Arduino-FFB-gui #3: setup

jackconcu123, 2022-06-02:
setup will now try to find control IO instances

ranenbg, 2022-06-02:
If it gets stuck at any of those steps, you can try and run the wheel control source filea from processing. I've put instruction for how to do that in a readme of github.

ranenbg, 2022-06-10:
The most often reason to get stuck is that the game controls library falsely tries to initialize devices like the mouse touchpad, Bluetooth or wifi as a game device and ofc fails. The workaround is when you run my program directly from processing IDE, it is able to pass normally in some case, so it's worth a try.

ranenbg, 2022-07-26:
I'm closing this, because it is actually not an issue, it's a feature of the app, guiding the user through first time setup process of the app. By default config file does not exist and you need to run app once in order to create it, after it does not ask for it anymore.

closed 2022-07-26 by ranenbg