ranenbg/Arduino-FFB-gui #5: I have a "manual" issue

tomascarabajal7, 2023-05-26:
Hi! 
I have a problem firstly understanding the manual. It makes mention on changing stuff like ppr. But it never tells you HOW or WHERE. 
And then im not able to configure mi pedals manually, only automatically. 
Thanks.

ranenbg, 2023-05-26:
Hi man, tnx for your questions. When you start my GUI you can see the blue number box field labeled CPR. This is where you can input your encoder PPR (not the same as CPR) by clicking once left mouse click, then type in the number you want by keyboard, then pressing enter, Please note that CPR=4xPPR if your encoder is mounted directly on the wheel shaft. If you mounted encoder via some gear or belt reduction system, then you have to take into account also the pulley/gear ratio in the calculation. Let's say that you have for each 1 wheel turn something like 10 encoder shaft turns, this would make a pulley/gear ratio of 10/1, so in that case you would have CPR=4xPPRx10.

In order to configure the pedals manually (by means of manual calibration) you have to upload to your Arduino board the firmware or HEX which supports manual calibration. Since you also asked some questions in my other repository for firmware, I will answer this question there since it is not directly related to GUI (the software part). The GUI is capable of automatically recognizing the firmware HEX version and enable/disable the options that are available/unavailable for that particulair firmware, so it will not allow you to manually change pedal calibration if the firmware does not support it.

tomascarabajal7, 2023-05-26:
Thank you for such a quick response! And thank you for the useful answer. I was now able to find where i can change the ppr. 
Also i thought that automatic was WITH manual, not one or the other. All sorted out. 
Besides some starting things the UI is super clear and easy to use.
Thank you so much 👏🤩

ranenbg, 2023-05-27:
You're welcome, I'm glad that you were able to get the GUI working easily. I'm closing this issue.

closed 2023-06-02 by ranenbg