ranenbg/Arduino-FFB-gui #6: Doubt

Valdemir-DSW, 2023-06-01:
Hello, first of all congratulations for your project, I'm also doing my custom one for my cockpit with telemetry and other things, but I'm in the force feedback part and I'm using a library that leaves my entire arduino reading because it consumes a lot of processing, I saw that You left the files but I didn't quite understand which libraries you used to make the force feedback work on the Arduino and how I made it turn into a USB joystick

ranenbg, 2023-06-02:
Hi, thank you. You are asking this question in the wrong repository, as this one is only for the GUI part which is independent of what goes on in Arduino. I'll reply to this question here, but if you have more questions please ask them in the Arduino firmware repository: Arduino-FFB-wheel.

I'm not using any particular FFB library for Arduino that you can download. The magic behind the FFB part in my firmware is in the modified Arduino USB core files, which in the manuals I refer to as the "modified core". They are a result of a brute-force effort in low-level C programming by several people, starting with Petter Barret. This is an extremely complex topic that revolves around DirectInput API (an integral part of Microsoft DirectX since 1999), which I also do not understand fully, but I understand it enough to use it. If you want to compile my firmware you can have a look at the readme of the firmware repository in the section "How to compile the source code". I explained there what you need to do in order to get those USB core files to work in your Arduino sketch. Once you compile it without errors you are free to modify it in any way you want. But take note that it took me about 6 months to get it working back in 2017 when I started with this project and another 2-3 years of work on other FFB wheel-related stuff, before I felt that it is complete enough to be uploaded on GitHub for the first time.

My advice to you is not to try to reinvent the wheel, use what others did before you and upgrade or add the features you think are missing. The telemetry part you are talking about is a try from game developers to completely ditch the DirectInput and just send the FFB data over the serial port to Arduino in a very basic way. Ofc, this makes their life much easier, but if you ask me it is not the correct way to utilize the FFB.

Valdemir-DSW, 2023-06-02:
Thanks for the quick reply sorry to ask the question in an inappropriate place, I'm analyzing your programming and your FFB files are very similar to mine the problem even here is optimization, I commented that there was going to be telemetry but it has nothing to do with force feedback is just for you to understand my project. could you tell me if these FFB files are compatible with stm 32. Any other questions I will ask in another Github

ranenbg, 2023-06-02:
Ok, I see. No, it is not compatible with stm32. You may want to have a look at openffb firmware, it is open spurce ffb wheel firmware for stm32 f407 and it's absolutely awesome project. They even made an AC servo motor driver that goes along with it.

closed 2023-06-09 by ranenbg