ranenbg/Arduino-FFB-gui #7: Questions about the usage of GUI

hungtcs, 2023-07-02:
Hello @ranenbg

Thank you for contributing to such an impressive project. I have successfully flashed Arduino-FFB-wheel to my Arduino, and everything is working fine now. But when I tried to use this GUI, I encountered some problems. I have tried various methods on my own, but none of them have provided a solution. I hope to receive your assistance and would be extremely grateful for it.

Let me explain my hardware setup first. 
I am using an Arduino Leonardo board, and I have successfully flashed the firmware "brWheel_my.ino.leonardo_v220.hex" using XLoader. When I use WheelCheck.exe to check the hardware, it also responds successfully.

Then I started using this GUI software. 
First, I followed the prompts to select the port and then chose my Arduino Leonardo development board.
Below are screenshots of the current steps.

[image]

This is a bit different from step three as described in the [data/Wheel_control_step3.png](https://github.com/ranenbg/Arduino-FFB-gui/blob/master/data/Wheel_control_step3.png), 
but I believe it's an update to the previous UI, so I proceeded with the operation.

Next, I saw this interface, and it's where I got stuck. 
I can't access the main interface as described in the [README](https://github.com/ranenbg/Arduino-FFB-gui#screenshot-of-gui-v251), no matter what I try.

[image]

From the UI, this interface seems to configure the functions of corresponding buttons through wiring. 
However, no matter how I connect them, when I click the Verify button, I always receive an error message like 'VERIFY 30 errors found'. 
Clicking the USE button is the same; it prompts 'SAVE - abandoned' and then there is no response.

The above is the complete description. I hope to receive your assistance. 
Thank you.

ranenbg, 2023-07-02:
Hi, thank you very much for such a detailed description of the problem. However, there is a simple solution. You have to connect all the controls like this:
[image]
Make sure not to make any crossings and connect all the buttons. This picture is inside the data folder along with other pictures that show examples of the startup setup for my GUI. Hope it helps.

hungtcs, 2023-07-04:
@ranenbg 

Thank you for your response. 
Actually, I had already tried connecting all the paths, but there were still errors. 
However, my issue has already been resolved, and the reason was that I didn't connect the potentiometers of the pedals.
Despite the fact that the pedal parts I bought online haven't arrived yet, I can't wait to use my steering wheel in advance 😆.
In conclusion, everything is perfectly fine now. Thank you very much.

closed 2023-07-04 by hungtcs