ranenbg/Arduino-FFB-gui #9: black screen gui

madhack2142, 2024-06-21:
hello got this problem, black screen in wheel control, tried to create com_cfg with com number same result
tried to connect arduino leo to pc without usbhub, no luck 
tried different firmware, no changes
noticed that wheel_control create a lot of javaw.exe after launch in task manager 
[image]

ranenbg, 2024-06-21:
Hi, I see that the pink screen is empty, nothing is connected. You need to get this done once and it should start normaly. I have described the procedure in manual.txt, in data folder of wheel control. In that folder you can also find a png example of pink window with all connections.

Give it a go first and let me know if you can start wheel control.

madhack2142, 2024-06-22:
[image]
[image]

wires already checked, axis have intense shaking

this problem happens on my laptop and i have no idea why, on pc i manually write port com comcfg and it works normal : ffb. axis
i tried to install new java 21 & 8

ranenbg, 2024-06-22:
Hi, it seems that you did not understand me. You keep missing a vital setup step. The images you posted again, the pink window is not completed, as there are no lines connected. You have clicked verify and it gave you 30 errors, because you didn't make a single connection between table on the left and the table on the right. Without this, wheel control setup can't complete. You have to do it once on every PC.

madhack2142, 2024-06-25:
i managed to solve my issue, tried arduino leonardo/promicro, got 24 or 25 errors when i have connected all buttons in software, a bit later decided to switch to brWheel_my.ino.leonardo_v230.hex without any letters at end and after connecting buttons it works flawlessly. noticed that on main pc there is no setup window step with buttons, application immediately crash after pressing OK, need to add device to comcfg manually and it starts normally

closed 2024-06-25 by madhack2142

ranenbg, 2024-06-26:
Hi, I'm glad you managed to sort it out.

Just please, make sure to use the latest wheel control GUI v2.6.0. You need it on order to utilize the full feature support of Arduino firmware HEX v230.