ranenbg/Arduino-FFB-gui #11: Screen stuck after completing configuration

N4TU5, 2024-09-07:
Hello. When I launch the wheel_control app on the bottom left for quite some time it says "Configuring wheel control". Then after couple of minutes it changes to "" and stays like it without loading anything else. I have tried re-downloading it from github but it didn't help.
If I try to run it manually using processing I get errors "frame cannot be resolved to a variable" in multiple files. What could cause this and what can I do to try and resolve it?
[image]
[image]

N4TU5, 2024-09-07:
Update:
I'm having problem with v232ZHX. I just tried with v230... and everything works, it launches quickly and without any issues. However I want to use load cell so I would like to use v232 :/

ranenbg, 2024-09-08:
Hi, tnx for reporting this issue, I see it for the first time. 

From info that you provided, it seems like something might be wrong with that particulair HEX version. For some reason, an empty string is returned as a response where GUI expects to read firmware configuration parameters, after sending command "U" to it. Do you have HX711 connected to arduino? You don't need to have load cell connected to HX711 at this point, but this chip has to be connected correctly to arduino, or the firmware may get stuck while trying to communicate with it. When you plug arduino to USB port, give it some time before you start GUI, because firmware for loadcell needs to do tare zeroing which takes 2 seconds. It is not required to have encoder connected at this point.

For every new firmware release, I'm manualy compiling 30+ HEX files, where I set by hand all options and version string. It's possible that I made some mistake with that particulair one. I'll have to check it myself in more details when I get some free time. If everything is fine with arduino, hx711 and wiring, it could be that I just need to recompile that HEX again.

In the meantime, if you wish to help, you may also do some troubleshooting using Arduino IDE and serial monitor. Set 115200 baud rate and carriage return as string terminator, then send command V, response sould be v232zhx. After that, try command U, response should be several numbers. You can check this with v230 first as that one works normaly. You can find a full list of all available commands in RS232commands_info.txt file.

N4TU5, 2024-09-08:
Hey. Thank you for the response! That's on me, I didn't think you need to have HX711 connected on startup, as it does work without any other inputs connected (encoder, potentiometers, buttons). I only tried it with HX711 once and now after first check with my multimeter it looks like there are some problems with my HX711 wiring. I will attempt to fix the wiring and try it again.

I'm closing this issue for now as I'm guessing fixing the wiring should resolve my problems

closed 2024-09-08 by N4TU5