ranenbg/Arduino-FFB-gui #13: Axis not changing when i rotate my wheel

JohnSK-dev, 2024-11-02:
Hello, 
I tried to install the Arduino-FFB-gui onto my PC and Arduino-FFB onto my arduino leonardo. The installation onto the arduino went fine, but when i tried to connect it to the GUI app, the sliders dont respond to the movement of my wheel, but the arduino seems to be recognized in the GUI software (I use leonardo_v170.hex on my arduino). Is there anything that i missed? Could you please help me resolve this issue? Aditionally when I open the DXTweak software the arduino seems to change the sliders as expected.

ranenbg, 2024-11-02:
Hi, I strongly recommend you to use the lattest firmware v240x and GUI 2.6.1 for the optimal results and problem free operation. You can use manual calibration feature to setup your pedals.

I did mention that lattest gui is backwards compatible all the way down to v170 HEX. But to be honest, I'm no longer checking if that's truly the case. There are so many HEX after a few years of developement and issue you're seeing may be due to some incompatibility of new GUI with old firmware. Just keep everything up to date and all should be fine.

JohnSK-dev, 2024-11-02:
I tired the v240ax firmware, but the sliders still dont do anything and I cant use the automatic COM port detection, because the GUI crashes whenever I try running it. But that is not a problem, because i create COM_conf.txt manually and the program turns on, but the problem with sliders may be because the GUI program doesnt show me the binding screen / setup. I dont know if this screen is needed or not, but it was said in the manual and i saw it on some photos. I dont know where the problem may be anymore. I am attaching the binding setup photo from the repository so you know what im talking about. 
[image]

ranenbg, 2024-11-02:
Do you have the pots correctly connected to analog inputs on Arduino? The pink screen is not required.

ranenbg, 2024-11-02:
Yeah, it does seem like there is a problem with axis/button mappings in GUI. It needs to be reset, but I don't know how, as this is part of game controls plus library that I'm using.

JohnSK-dev, 2024-11-02:
Is this the menu you are referring to? If so, how can i open it ?
[image]

ranenbg, 2024-11-02:
Yes, it should open when you delete com_cfg from data folder of gui. If the gui crashes before you can complete the setup, it's a bummer as I can reproduce those conditions on my PC and can't do anything about it.

JohnSK-dev, 2024-11-02:
I did it and now this is where the program stops 
[image]

ranenbg, 2024-11-02:
Ok, I think I can see where the issue is, tnx for sending this to me. 

So in data folder of gui you will find the button/axis bindings inside the "Leonardo wheel v5" file. You need to place it in the folder where it is missing - this folder where terminal is saying, first red line.

JohnSK-dev, 2024-11-02:
OK i did it and now there is some other issue: 
[image]

ranenbg, 2024-11-02:
delete com_cfg from data folder and try again

JohnSK-dev, 2024-11-02:
Yeah it keeps failing with the same error and the exe file still doesnt work, Im gonna try giving it the COM port manually

JohnSK-dev, 2024-11-02:
The same problem as before, it doesnt update the rotation and the pedal sliders dont work also
[image]

ranenbg, 2024-11-02:
Oh man, I see where the issue is. Everything was fine with gui all along, we were sidetracked.

The problem was created because you probably had some other code running in that Arduino, which wrote some registers in EEPROM. Those hold calibration values for pedals. I can see that YR command retrives values 65535 which is max out 16bit number. This has to be fixed. Here's what you need to do.

JohnSK-dev, 2024-11-02:
Thank you, the sliders move now, it was a VJoy conflict, when i deleted VJoy, the steering wheel started working as expected. I can confirm that the arduino was at COM 12 and the manual pedal setup worked.
[image]
[image]

ranenbg, 2024-11-02:
I'm glad that you managed to solve it by uninstalling vJoy, you don't need that at all.

Have a look at github readme for gui repository on how to solve incorrect X-axis calibration, where it gets stuck at 540deg.

JohnSK-dev, 2024-11-23:
Hello,
I was wondering if there was any way of playing Forza Horizon 5 with a steering wheel set up this way? Because I already had my wheel set up another way before and i used Forza Emuwheel to make it happen, but this needs Vjoy to function and there is a problem that Vjoy breaks Arduino-FFB-Gui programs inputs. So I dont know what to do now, or if there is any different way of doing this. Any help with this will be appreciated.

ranenbg, 2024-11-24:
Hi, please in future make a new issue if you have a new question and close existing thread if I helped ypu to solve the original problem.

Forza Horizon 5 does not supoort direcInput game controlers. It uses newer xInput protocol and the only way to get it to work is to emulate the DI device into an XI one using various 3rd party software like emuwheel.. You can give a try this:
https://github.com/csutorasa/XOutput