ranenbg/Arduino-FFB-gui #14: everything is gray and more profiles

Notriatv, 2024-11-03:
hi again XD 
1. so. while simhub is open, even in background, wheel control don't show nothing..
2. is it possible to add more than 10 profiles? bc I have 4 just for assetto corsa lol and many other games XD
3. thank you again

ranenbg, 2024-11-03:
Hi, I think that simHub is taking over serial ports and that is why my GUI can't get acces to arduino com port. Not much I can do about that unfortunately.

Yes, I can add more profiles, these are stored in PC so there is no memory limitation. Will do it in new GUI release, tnx for suggestion.

Notriatv, 2024-11-04:
Ah yes, that's true, you're right. The solution is simple: I just uncheck the Arduino with wheel firmware in the Arduino scan settings in SimHub, and now the GUI opens without needing to close SimHub. Thanks for the tip

[image]

ranenbg, 2024-11-04:
Great, thank you this will be very useful for others also.

I'll close this issue as it is solved.

closed 2024-11-04 by ranenbg