ranenbg/Arduino-FFB-gui #16: Black window with "Configuring wheel control" message

Pancholina, 2024-11-26:
First of all I would like to thak you all the work done and the support you are providing to everybody.
As I'm trying to use a diy F1 replica wheel, I needed to include the clutch reading value in the data stream received by the shift registers from the wheel. At the end the HID data arrangement is like you described in "input_report_bytes_ver3.png" at the "wirings" directory, using  the "Joystick.send_16_16_12_12_32(m_x,m_y,m_z,m_rx, m_buttons)".
Once the firmware compiled, with all the modifications, I tried to configure connecting just the Leonardo board, with "Wheel control" program. 
I removed the two lines related with "RYaxis" and  "Hat switch" in the "Arduino Leonardo wheel v5" file to avoid errors during configuration (because I should left them unconnected), and after connecting all the signal during the configuration phase, I only have the black window with the "Configuring wheel control" message.
I don't know if this problem is due to the fact that I have only the Leonardo board connected (without any other device), or because there is something wrong.
I have tested the Leonardo board with this firmware using all the other programs you collected in the "FFB_misc_programs" and looks that everything is OK.
Where could be the problem? What is happening to block the program in this black window?

ranenbg, 2024-11-26:
Hi, thank you, you are welcome.

There could be a number of things causing the problem. Before we go any firther, what firmware version are you trying to modify? Did you try to run GUI as a source from processing? You will have exact issues listed in terminal that are causing the program to get stuck.

ranenbg, 2024-11-26:
Good, but if you want my I help I need the info from terminal to see why gui gets stuck. Processing IDE has a very similair language as Arduino IDE, ypu'll manage.  I also need to know exact changes you did to nano and firmware

ranenbg, 2024-11-26:
You can send me the modified source files of firmware on email, tnx.

Pancholina, 2024-11-26:
Hi.
I was trying with the precessing IDE but I have "The package “org.gamecontrolplus” does not exist. You might be missing a library.
" message.
Any idea?

Pancholina, 2024-11-26:
Please let me know at which email address ca I send to you the modifications.

ranenbg, 2024-11-26:
It's in wheel control, on the bottom where it says info about versions.

Pancholina, 2024-11-27:
I was trying to run wheel_control through the processing IDE, and after installing all the libraries I had some error messages.
I saw that the link for installing processing IDE is pointing to download processing 4.3 IDE
After installing processing 3.5.4 IDE (as suggested) the program is running as the ".exe" version, and now I can check where it stops.
If I remove the line related with the hat switch (in "Arduino Leonardo wheel v5" text file) everything is running well, but if I remove the line related with RYaxis the program stops when trying to read this channel.
[image]
As I'm not using RYaxis I cannot connect it during initial setup and the configuration cannot be finished. But If I remove this line (as I did with the hat switch) the problem stops.
So, how to proceed when I don't need the hand brake?

ranenbg, 2024-11-27:
Ok, I see. Yes, this error is expected, because my version of GUI needs all axis that are defined in firmware in HID.cpp, and those are also correspondingly defined in hid-to-gui mapping file v5. 

I'm not sure why you feel the need that anything needs to be removed. Simply in firmware put 0 value for all axis you do not plan to use and you don't need to change anything in gui.

Pancholina, 2024-11-27:
Hi Milos,
Understood. My mistake was to assume that not only the "ver5" could be used, and modifying the mapping file every configuration described inHID.cpp could be valid.

In my case I was using what you called "ver3":

// milos ver3, 16+16+12+12 bits version + 32 buttons
void Joystick_::send_16_16_12_12_32(int16_t x, uint16_t y, uint16_t z, uint16_t rx, uint32_t buttons)
{
  // milos, total of 11 bytes, 2B for x, 2B for y, 3B for z, rx, 4B for buttons
  u8 j[11];
  j[0] = x;
  j[1] = x >> 8;
  j[2] = y;
  j[3] = y >> 8;
  j[4] = z;
  j[5] = (z >> 8) & 0xf | ((rx & 0xf) << 4);
  j[6] = rx >> 4;
  j[7] = buttons;
  j[8] = buttons >> 8;
  j[9] = buttons >> 16;
  j[10] = buttons >> 24;

  HID_SendReport(4, j, 11);
}

And, as you suggested, the only one that can be used is "ver5":

// milos ver5, 16+16+12+12+12+12 bits version + 32 buttons
void Joystick_::send_16_16_12_12_12_12_32(int16_t x, uint16_t y, uint16_t z, uint16_t rx, uint16_t ry, uint16_t rz, uint32_t buttons)
{
  // milos, total of 14 bytes, 2B for x, 2B for y, 3B for z and rx, 3B for ry and rz, 4B for buttons
  u8 j[14];
  j[0] = x;
  j[1] = x >> 8;
  j[2] = y;
  j[3] = y >> 8;
  j[4] = z;
  j[5] = (z >> 8) & 0xf | ((rx & 0xf) << 4);
  j[6] = rx >> 4;
  j[7] = ry;
  j[8] = (ry >> 8) & 0xf | ((rz & 0xf) << 4);
  j[9] = rz >> 4;
  j[10] = buttons;
  j[11] = buttons >> 8;
  j[12] = buttons >> 16;
  j[13] = buttons >> 24;

  HID_SendReport(4, j, 14);
}

Thank you for the clarification.

ranenbg, 2024-11-27:
Yes, exactly, ver3 input report function for HID input report was used along side older version of GUI, which is no longer compatible. After that I made ver5, that goes along with "v5" mapping for gui.

Pancholina, 2024-12-19:
Hi again.
I don't know if it would be better to open another issue, or follow with this one.
Anyway, I was following with the project and now it looks that everything is running, but I have faced an unexpected result, and now I don't know if I have now some unexpected side effects after my firmware modifications or may be, the wheel is working at it should.
I was expecting that enabling the "centering" option, I should have a FFB response as this:
[image]
But instead of this response I have this:
[image]
I have tried playing with all the configuration parameters but I haven't seen any improvement.
My questions are: 
Are my expectations wright?
Should I have the linear response of the first graph?
If no, how can I change this?

Thanks in advance.

ranenbg, 2024-12-19:
Hi, centering only sets the wheel angle at 0deg. The setting that you're looking for is called FFB deadzone and it doesn't exist in my firmware as this is not needed. However, what you need in order to compensate for inherent FFB deadzone due to friction in the wheel system is minimal torque and that setting is available. You can play around with min torque slider. You should find balance between minimal FFB deadzone when using auto-centering spring and wheel oscilations.

Pancholina, 2024-12-19:
Hi.
May be I was not explaining well.
What I was expected was just to recenter at 0deg but I found a wide deadzone that I was not expecting.
My wheel now is not going back to the center (0deg).

ranenbg, 2024-12-19:
It can't go back perfectly to the center with the auto-centering spring effect enabled, you should not be bothered by that. The idea behind min torque is that you can feel low forces, since they are eaten away by friction in the system.

Pancholina, 2024-12-19:
Understood.

Thank you.

closed 2026-02-01 by ranenbg