ranenbg/Arduino-FFB-gui #18: build issue

neas15us, 2025-02-25:
im trying to build the source and i keep getting:
frame cannot be resolved to a varible

trying to build it for linux

ranenbg, 2025-02-25:
Hi, linux is not supported. Even if you upload a compiled HEX, it will probably not report it as a working device.

ranenbg, 2025-04-05:
If you're talking about trying to build it in linux, I'm sorry I can't give any help because I have 0 knolewgde about it.

MGsBueno, 2025-11-13:
Hi, I'm having the same building issue on Windows 11.
When i try to run the GUI it closes after the step configuration

ranenbg, 2025-11-13:
Hi, as I wrote in the instructions, for Windows, you have to use Processing version 3.5.4 and not 4 which gives this error for some reason.