ranenbg/Arduino-FFB-gui #19: GUI size

pecedevelopment, 2025-04-06:
Hi!

Huge respect for your work. I just started using your software (attached rotary encoder to arduino), in overall looks very promising, however it would be very great if the window size could be resizeable. I have a (small) laptop screen with 2k resolution and it was really hard to read the UI. 

Have a nice day!

From a first year electrical engineer student :)

ranenbg, 2025-04-07:
Hi, thank you. It is indeed a cool suggestion for a user perspective, but a nightmare for any dev.

This is something I wanted to do a long time ago and that is why for the most elements you see in GUI I made my own classes from scratch for which I could set the size an position of each element. I did have resizing in mind from the very beginning, but I settled for a fixed 1440×900 window resolution and did my best to size and place everything in a well aranged manner where nothing gets blurry.

Converting GUI to a dynamicly resizeable window with an arbitrary aspect ratio and having everything looking the same just scale up in size while keeping the relative positions is an enourmous task, due to large number of possibilities that all have to be taken care of.

Even if I could pull it off somehow, there is a pink configuration window that would stay the same size (it's someone else's library), so as the java popup windows for user input, questions and information. This would simply look comically small (or large depending on main window size) and I couldn't do anything about it.

However, I'll keep thinking about this and maybe I'll come up with something useable for bigger screen resolutions. Something like a pre selected gui size in a config text file where one could choose between 1×, 1.5× or 2× seems doable, but I'm not promising anything.

pecedevelopment, 2025-04-07:
What do you think about scaling to fixed windows? I mean we create gui for 13??x 7??, fullhd, etc. for standard screen size

pecedevelopment, 2025-04-09:
what versions are you using for the following libaries?

    Game Control Plus
    G4P
    Sprites
    ControlP5

ranenbg, 2025-04-09:
The latest.

ranenbg, 2025-04-09:
Please read the github readme, I explained how to compile the source.

pecedevelopment, 2025-04-09:
I already did that before writing my last comment :/

ranenbg, 2025-04-09:
You need to have all pde files opened as a separate tab in a single project. You can do that if you open wheel_control.pde and have all other files in wheel_control folder.