ranenbg/Arduino-FFB-gui #20: application closes by itself

Mrbraule, 2025-05-28:
When I connect the lines, I check and when I use it the app closes!!

Note: I used it without any problems because I formatted my PC and when I tried to use it, this problem started to appear.
Does the app need any program or resource to work, such as netframework, etc.?

ranenbg, 2025-05-28:
Hi, the app needs the latest Java and you may be missing it on your fresh Windows install. But I'm suspecting something else is the reason for the app to close. It would help if you post the contents of your log, there may be some info that can narrow down the issue, it's in data folder.

Mrbraule, 2025-05-28:
is this the file?

[guiChangeLog.txt](https://github.com/user-attachments/files/20489007/guiChangeLog.txt)

> Olá, o aplicativo precisa da versão mais recente do Java e pode não estar disponível na sua instalação recente do Windows. Mas suspeito que haja outro motivo para o aplicativo fechar. Seria útil se você publicasse o conteúdo do seu arquivo setupTextLog, pois pode haver alguma informação que possa ajudar a solucionar o problema. Ele está na pasta de dados.
> 
> Normalmente, na GUI v2.6.5, fica algo assim (dependendo da versão do firmware):
> 
> Configuração do controle de roda v2.6.5 inicializada Resolução: 1440x800 Dispositivo HID: Arduino Leonardo Porta: COM5, carregada do txt Cores do eixo: carregadas do txt Configurações de firmware detectadas 1080 100 50 50 100 100 100 50 70 100 0 128 1 2047 4000 12 Versão HEX: fw-v250fx Shifter analógico XY H detectado 255 511 767 255 511 0 Eixo analógico para FFB habilitado Calibração manual para pedais habilitada 0 1023 0 1023 0 1023 0 1023 Usando EEPROM para carregar/salvar configurações Configuração concluída
> 
> Ainda estou trabalhando na finalização da GUI v2.6.5, mas o log da v2.6.4 também deve ser suficiente. Se não fornecer nada útil, é melhor instalar o processamento 3.5.4 e executar o aplicativo como código-fonte. O terminal de status mostrará um erro detalhado que certamente identificará o motivo.

is this the file?

Mrbraule, 2025-05-28:
> Não, isso não está correto. Preciso do seu setupTextLog.txt

[setupTextLog.txt](https://github.com/user-attachments/files/20491078/setupTextLog.txt)

Mrbraule, 2025-05-28:
> Este é o log criado no meu PC que enviei para o GitHub. Ele não nos diz nada sobre o seu problema.
> 
> O Arduino é visível em jogos como um joystick ou um controlador de jogo? Certifique-se de que todos os dispositivos i2C externos estejam conectados ao Arduino se você usar um firmware HEX compatível. Caso contrário, o Arduino pode travar ao tentar se comunicar com dispositivos inexistentes.
> 
> Verifique no gerenciador de dispositivos se o número da porta COM do Arduino está escrito corretamente em COM_cfg.txt e tente executar a interface gráfica como administrador. Tente também excluir o arquivo COM_cfg.txt para redefinir o procedimento inicial de configuração da interface gráfica. Siga atentamente cada passo e envie-me uma captura de tela quando chegar a uma etapa em que todos os dispositivos de jogo reconhecidos estejam listados. Talvez a interface gráfica trave ao tentar configurar incorretamente o touchpad ou o dispositivo Bluetooth como um controlador de jogo se você estiver usando um laptop.

where can i find COM.cfg?

Mrbraule, 2025-05-29:
> COM_cfg.

So I did it and it worked!
I really appreciate the support.
I created the cfg file and it worked, God bless you and thank you