ranenbg/Arduino-FFB-gui #22: App crashes and error during compiling

Abhay-Maurya, 2025-09-07:
when I try to run gui it crashes after available devices window.
and if I try to compile all the pde file in processing IDE, I get following error

[image]

Abhay-Maurya, 2025-09-08:
I compiled source code in Processing 3.5.4 and the problem seems to be Arduino itself.
when i disconnect arduino the program doesn't crashes

ranenbg, 2025-09-08:
Ok good, now you sorted everything out, it seems to be working fine.

You have to put any unused analog input to GND, or it will pickup the noise if left floating (unconnected), it's normal. Pots themselves are not noisy, it's the unshielded wiring, which picks up noise. Flickering of about +-1 out of 1023 is normal arduino ADC noise, anything above that is due to bad wiring. You can use audio stereo cables, use shield as ground.

Abhay-Maurya, 2025-09-08:
Alright, so everything was fine, steering wheel working, pots working, but as soon as i turned on motor for FFB the pots went crazy, fluctuating in 0-100 range and steering wheel started acting a little bit jittery.

is there a manual for all the gain and setting in wheel control for understanding what all those gain % do?

ranenbg, 2025-09-09:
RC network is just a filter that smooths out voltage from a pot, by removing high-frequency content, but it does nothing to prevent EMI.

For protection against back EMF from the motor, you need to build some form of braking resistor circuit. There are many variants of these from simple passive, to active ones, but it goes out of the scope of my project. A capacitor alone is definitely not one of such circuits.

Abhay-Maurya, 2025-09-09:
yeah, original issue resolved, thank you very much. closing this thread now